knurling-rs/defmt #850: Compatibility with PlotJuggler

asasine, 2024-06-06:
This might seem like a longshot but how would I go about supporting defmt logs in [PlotJuggler](https://github.com/facontidavide/PlotJuggler)? My idea is to bring the functionality similar to Arduino's Serial Plotter to defmt-based systems such that recurring logs with the same format, but differing values for a certain variable, can be plotted. I think PlotJuggler is a good choice as it's an incredibly powerful tool, already has plugin support, and is transport- and protocol-agnostic. Building upon that brings a lot of value to defmt-based systems without needing to build the entire plotting application from scratch.

Urhengulas, 2024-06-10:
`probe-run` did support a `--json` mode which might be handy here. You could build a script/tool/plugin the reads the json and feeds it into PlotJuggler. You can read about it here: https://defmt.ferrous-systems.com/json-output

The problem is that `probe-rs` does not support the json mode atm, but it should not be too hard to add it, since all the formatting logic is in `defmt-decoder`.

avsaase, 2025-05-30:
> What would you want it to look like if defmt-decoder could give you 'raw' values?

In the simplest form it could output the log message with placeholders for the values and an array of values and types like this:

```json
{
    "message": "read_mag: Read raw mag data: MagData(Matrix { data: ArrayStorage([[{0}, {1}, {2}]]), _phantoms: PhantomData })",
    "data": [
        {
            "value": -2472.0,
            "type": "f32"
        },
        {
            "value": 3855.0,
            "type": "f32"
        },
        {
            "value": 3297.0,
            "type": "f32"
        }
    ]
}
```

Ideally it would describe the whole nested message structure but I think the message is already pasted together at firmware compile time so that won't work.