knurling-rs/defmt #851: Online defmt book needs to be updated

plaes, 2024-06-10:
Currently published [defmt book](https://defmt.ferrous-systems.com/) seems to be seriously outdated as its version is: `commit e1fd1e6 on 2024-02-07 19:08:44`.

Urhengulas, 2024-06-10:
On it

andershedberg, 2024-10-25:
I tried to look in the document to see how the logs actually look, and that is sadly missing. No examples at all of logfiles.

Urhengulas, 2024-10-25:
> I tried to look in the document to see how the logs actually look, and that is sadly missing. No examples at all of logfiles.

In the display hint chapter you can see the output as comments in the code: https://defmt.ferrous-systems.com/hints.

Urhengulas, 2024-11-06:
We should switch from netlify to github pages using https://github.com/actions/upload-pages-artifact and https://github.com/actions/deploy-pages.

Urhengulas, 2024-12-02:
The defmt book is now hosted on cloudflare and is tracking the main branch.

closed 2024-12-02 by Urhengulas