knurling-rs/defmt #853: Defmt's proc macros issue an error if the filename is not a correct identifier

gregtwice, 2024-06-11:
I have a file name 00_Hello_world in my bin folder which contains the following code:

```rust
#![no_std]
#![no_main]

use embassy_executor::Spawner;
use embassy_stm32::gpio::{Level, Output, Speed};
use embassy_time::Timer;
use {defmt_rtt as _, panic_probe as _};

use defmt::*;

#[embassy_executor::main]
async fn main(_spawner: Spawner) {
    let p = embassy_stm32::init(Default::default());

    let mut led = Output::new(p.PA5, Level::High, Speed::Low);

    loop {
        info!("{}", 2);
        led.set_high();
        Timer::after_millis(300).await;

        // defmt::info!("low");
        led.set_low();
        Timer::after_millis(300).await;
    }
}
```
When compiling I get the following error:
``` 
error: `{input}` is not a valid identifier
   --> src\bin\00_Hello_world.rs:18:9
    |
18  |         info!("{}", 2);
    |         ^^^^^^^^^^^^^^ in this macro invocation
    |
   ::: C:\Users\gregtwice\.cargo\registry\src\index.crates.io-6f17d22bba15001f\defmt-macros-0.3.9\src/lib.rs:137:1
    |
137 | #[proc_macro_error]
    | ------------------- in this expansion of `info!`

```
It compiles fine when removing the "00_" part of the file, or when removing the macros. So I guess that the issue is due to the fact that my filename is not a correct identifier. Do you think the error message could be improved? Or is it outside the scope of the crate?

Thank you for your consideration.

Urhengulas, 2024-06-12:
The error stems from https://github.com/knurling-rs/defmt/blob/75fb317dab468a9fc786fe1f1da421b6b24e64d2/macros/src/function_like/log/env_filter/parse.rs#L102-L105

lure23, 2024-08-09:
Bit by this as well, on `examples/1-ranging-basic.rs` etc.

Would appreciate the limitation to be lifted if there's no technical reason for its existance.

Urhengulas, 2024-08-14:
@lure23 @gregtwice I gave it a shot in https://github.com/knurling-rs/defmt/tree/accept-crate-names-with-leading-numbers, but could not get it to work fully. Feel free to pick up the work.

lure23, 2024-08-14:
Unfortunately I’m not confident enough yet, in Rust, but thanks for the link. Prefixing with `_` serves me well for now.

Urhengulas, 2024-10-18:
> Unfortunately I’m not confident enough yet, in Rust, but thanks for the link. Prefixing with `_` serves me well for now.

Probably that would be enough to fix it, just internally prefixing it with an underscore.

Urhengulas, 2024-11-06:
It would be good to know (and document!) why the path segments even need to be valid identifier. Maybe we can even lift it?

Briefly looking in the `env_filter` implementation (https://github.com/rust-cli/env_logger/blob/main/crates/env_filter/src/parser.rs#L58) it seems like they do not place this restriction upon their path segments. But our context is a bit different since our filter runs during compiletime.

gregtwice, 2025-02-24:
> [@lure23](https://github.com/lure23) [@gregtwice](https://github.com/gregtwice) I gave it a shot in https://github.com/knurling-rs/defmt/tree/accept-crate-names-with-leading-numbers, but could not get it to work fully. Feel free to pick up the work.

@Urhengulas What is missing in your implementation? I happen to find myself with a bit of free time, I could take a look at it.