knurling-rs/defmt #854: Provide mechanism to export defmt mappings from the ELF

ryan-summers, 2024-06-13:
There are a number of potential uses of `defmt` that don't actually require the ELF binary, such as:
* Using `defmt` in device logs for compact storage
* Using `defmt` to generate telemetry information

Some external receiver is then responsible for decoding and formatting these values at a later time/date. Currently, this process requires an `elf` binary file that generated the `defmt` logs in the first place.

Being able to export the `defmt` mapping tables into some other format (i.e. TOML, JSON, YAML, etc.) would aid in the ability to write custom tools to process these logs after-the-fact.

jonathanpallant, 2024-06-20:
That sounds really interesting.

Is the symbol table sufficient (apologies, I should dig in and check to answer that, but I haven't yet)? IIRC the defmt strings are all JSON encoded symbol names that are easy to find from an `objdump`.

reinismu, 2025-10-17:
@ryan-summers did you ever solve this?

I also see huge potential of using defmt logging on desktop and server applications. It would reduce the network bandwith a lot for streaming logs.
Maybe there are other libs specificly for that, but I haven't seend any