knurling-rs/defmt #861: Object safe Format

EdBuilds, 2024-08-18:
Hello, 
Would it be possible to make the `Format` trait object safe? I would like to pass messages around by using 
```
&dyn defmt::Format 
```
But the static function in the trait prevents that.
Cheers!

jonathanpallant, 2024-11-05:
I looked into this for you.

The serialiser needs to report what kind of *thing* is being serialised so the deserialiser can unpack it. This is done using an interned string:

```rust=
pub fn fmt<T: Format + ?Sized>(f: &T) {
    istr(&T::_format_tag());
    f._format_data();
}

pub fn fmt_slice<T: Format>(values: &[T]) {
    usize(&values.len());
    istr(&T::_format_tag());
    for value in values {
        value._format_data();
    }
}
```

If `_format_tag()` takes `&self`, then you can't call `T::_format_tag()` when serialising a slice, and we can't guarantee that the slice has non-zero length, so we can't call `values[0]._format_tag()` either. So the trait has to include a static method, and that means we can't have dynamic trait objects.

So with the current design, no, it's not possible to have a `&dyn defmt::Format`. Can you elaborate on why you would like to make `&dyn` references to things that can be formatted with defmt? What would you wish to do with those reference?

jordens, 2024-11-22:
One could change the slice format and omit the inner format tag in case of an empty slice. The data is already there to recognize it on the decoder (zero length). And it would save at least one byte. Then `_format_tag()` can become a regular method.

jonathanpallant, 2024-11-26:
Ah.

```rust
pub fn fmt_slice<T: Format>(values: &[T]) {
    if let Some((first, remainder)) = values.split_first() {
        usize(&values.len());
        istr(&first.format_tag());
        first._format_data();
        for value in remainder {
            value._format_data();
        } 
    } else {
        usize(0);
    }
}
```

That would be a breaking change to the encoding, which would need a wire format version bump. But it perhaps it is worth it.

thejpster, 2025-01-29:
We discussed this on the Matrix chat. If you'd like to propose a PR, we can look at exactly how breaking a change it is.

thejpster, 2025-01-29:
The trick is handling the difference between a slice of T (where the format is interned once) and a slice of &dyn Format, where you need to intern the format for each item in the slice. It wasn't immediately clear how to do that.

jonathanpallant, 2025-01-29:
We discussed this on the Matrix chat. If you'd like to propose a PR, we can look at exactly how breaking a change it is.

The trick is handling the difference between a slice of T (where the format is interned once) and a slice of &dyn Format, where you need to intern the format for each item in the slice. It wasn't immediately clear how to do that.