knurling-rs/defmt #862: Compatibility issue with rust 1.82.

kareled, 2024-08-26:
Hello,
not sure where to log it, but with update to 1.82 nightly my project link fails with missing ` __defmt_default_panic␍`

The project builds for armv7emhf and it links fine code unchanged with various rustc toolchains up to including 1.81 nightly. Always using nightly due to RTIC usage. With nightly from 2024-08-03 (my first obtained nightly of 1.82 family) it starts to complain and fail while being linked with:
```
  = note: rust-lld: error: C:\Users\gardas\git\develop\target\thumbv7em-none-eabihf\debug\build\defmt-e20cffcd2d13aae7\out\defmt.x:7: symbol not found: __defmt_default_panic␍
          rust-lld: error: C:\Users\gardas\git\develop\target\thumbv7em-none-eabihf\debug\build\defmt-e20cffcd2d13aae7\out\defmt.x:7: symbol not found: __defmt_default_panic␍
          rust-lld: error: C:\Users\gardas\git\develop\target\thumbv7em-none-eabihf\debug\build\defmt-e20cffcd2d13aae7\out\defmt.x:7: symbol not found: __defmt_default_panic␍
          rust-lld: error: C:\Users\gardas\git\develop\target\thumbv7em-none-eabihf\debug\build\defmt-e20cffcd2d13aae7\out\defmt.x:7: symbol not found: __defmt_default_panic␍

          flip-link: the native linker failed to link the program normally; please check your project configuration and linker scripts
```
If this is not defmt issue, please recommend where to report it while closing. Thanks!

skade, 2024-10-08:
This is due to upstream bug https://github.com/rust-lang/rust/issues/131164

Urhengulas, 2024-10-09:
@Dirbaio [said](https://github.com/rust-lang/rust/issues/131164#issuecomment-2400724853):
> A possible workaround defmt could do is adding EXTERN(_defmt_panic) to the linker script. The side effect is it'd prevent the panic handler from being optimized out when it's truly unused, I think.

I can confirm this works as a workaround on current beta. Let's hope it won't be necessary, but if the upstream fix does not make it into stable we can use it.

Urhengulas, 2024-10-09:
It probably should be `EXTERN(__defmt_default_panic)` instead of `EXTERN(_defmt_panic)`

Dirbaio, 2024-10-09:
I think `EXTERN(_defmt_panic)` is better because if the user has defined their own panic handler, i'tll just force-keep that one, and the default one will be optimized out. If you do `EXTERN(__defmt_default_panic)` it'll force-keep the default one even if the user has overridden it.

(I haven't actually tested this, i'm just guessing based on my mental model of how links work which might be wrong because linkers are cursed)

Urhengulas, 2024-10-14:
Current beta contains the fix, so no action is necessary from our side. We will still have a look if adding the extern to the linker script makes sense in case the issue comes back.

jannic, 2024-10-16:
Stable pre-release (https://internals.rust-lang.org/t/rust-1-82-0-pre-release-testing/21705) also contains the fix.

Urhengulas, 2024-10-16:
Perfect. Thanks to everyone involved in managing the issue!

closed 2024-10-16 by Urhengulas