knurling-rs/defmt #863: defmt-macros depends on proc-macro-error which is considered unmaintained

ia0, 2024-09-06:
I'm using `defmt` and thus `defmt-macros`. I've recently got a security warning because I depend (through `defmt`) on `proc-macro-error`. See https://rustsec.org/advisories/RUSTSEC-2024-0370.html for more details. Would it be something you would consider to either migrate to `proc-macro-error2` or avoid depending on `proc-macro-error` through another mean?

Urhengulas, 2024-09-07:
Thank you for notifying us @ia0.

Either using `proc-macro-error2` or just using `syn::Result` seem like viable options to me.

Would you be interested in making either change?

ia0, 2024-09-08:
Thanks for the answer.

This is not a big issue on my side so far (unmaintained crates don't really have an urgency aspect as other security issues may have), so I won't be able to allocate time for it, sorry.

closed 2024-09-16 by Urhengulas