knurling-rs/defmt #867: feature req: defmt::expect! macro like unwrap!

dvdsk, 2024-09-27:
The `unwrap!` macro got implemented [see](https://github.com/knurling-rs/defmt/issues/161) , in that issue a `expect!` macro is also proposed and an implementation sketched out . Only the `unwrap!` macro got implemented however. Is there a reason for this?

To me an `expect!` macro would be very useful .

Urhengulas, 2024-09-30:
It would be useful indeed. Are you interested in implementing it?

dvdsk, 2024-09-30:
I'll have a go at it, not familiar with the internals of defmt however since there is a copypaste-ready implementation and I can have a look at `unwrap!` I can probably figure it out. Then we can hammer out the details in the PR

dvdsk, 2024-10-01:
So I had a quick look at noticed that `unwrap!` already implements the behavior of `expect!`. As in it allows a second argument that is a message. So `expect!` is implemented as `unwrap!`. See the docs here: https://docs.rs/defmt/latest/defmt/macro.unwrap.html

So this seems to be more an issue of discoverability then missing features. What to do there. I propose we make `expect` an alias for the current `unwrap`-macro. That way users that know of `unwrap!` can discover `expect!`. It might also be worth mentioning `unwrap!` in the defmt book. If I am not mistaken its not in there atm.

Urhengulas, 2024-10-09:
> So this seems to be more an issue of discoverability then missing features. What to do there. I propose we make `expect` an alias for the current `unwrap`-macro. That way users that know of `unwrap!` can discover `expect!`.

Agree

> It might also be worth mentioning `unwrap!` in the defmt book. If I am not mistaken its not in there atm.

and agree

dvdsk, 2024-10-09:
PR https://github.com/knurling-rs/defmt/pull/872 for the alias

closed 2024-12-02 by Urhengulas