knurling-rs/defmt #868: panic probe not resetting device

Gibbz, 2024-09-28:
I'm using panic probe with embassy rust. It's supposed to reset the device when there is a panic.
The logging part is working fine, however the device stays stuck after the log. There is no reset.

Is there some extra setting required to cause it to reset after the log?

Urhengulas, 2024-10-09:
Is it possible you could share your workaround? As a reference for others and maybe an implementation

Gibbz, 2024-10-10:
I used the following

'''rust
use core::panic::PanicInfo;
#[panic_handler]
fn panic(info: &PanicInfo) -> ! {
    cortex_m::interrupt::disable();
    defmt::error!("Panicked: {}", defmt::Display2Format(info));
    //panic_persist::report_panic_info(info);
    for _ in 0..2_000_000 { // delay before reset
        nop()
    }
    cortex_m::peripheral::SCB::sys_reset();
}
'''