knurling-rs/defmt #877: Allow bitflags version one and two to coexist

Urhengulas, 2024-10-18:
There is a `defmt::bitflags!` macro which depends on the `bitflags::bitflags!` macro from the `bitflags v1` crate. Since it's introduction there was a `bitflags v2`, but unfortunately we cannot just bump the dependency, since the `defmt::bitflags!` macro leaks implementation details of the `bitflags::bitflags!` macro and therefore the dependency update could break the code of defmt users.

Therefore we want to add two new cargo features to defmt: `bitflags1` and `bitflags2`. They each gate a `defmt::bitflags` macro which depends on `bitflags v1` and `bitflags v2` respectively. To not break any existing code, the `bitflags1` feature should be enabled by default. A `compile_error` when both are enabled sounds useful as well.

Note that there is an existing PR to update to bitflags v2: https://github.com/knurling-rs/defmt/pull/746.

Urhengulas, 2024-11-28:
True. So I propose the following:
- hide `defmt::bitflags` behind a feature but enable it by default
- add bitflags v2 as `defmt::bitflags2!` behind a feature, not enabled by default

@jonathanpallant do you see any problems with that approach?

cc @manushT

jonathanpallant, 2024-11-28:
Do we have any existing features and is anyone out there likely to be using defmt with `default-features = false`? Because that proposal would break them.

The other option is:

1. Leave `defmt::bitflags` as it is
2. Add a new feature `bitflagsv2` which provides `defmt::bitflagsv2` using an optional dependency.

It's still not unclear to be if anyone *needs* us to upgrade to bitflags v2. It's an improvement, but I don't think bitflags v1 is broken for anyone right now.

zheylmun, 2025-03-22:
> It's still not unclear to be if anyone _needs_ us to upgrade to bitflags v2. It's an improvement, but I don't think bitflags v1 is broken for anyone right now.

One note is that adding additional derives with the v1.x bitflags becomes unnecessarily tricky.
As an example It would be nice to be able to derive the `serde` traits on a bit flags type.

lynxD, 2025-05-20:
Is 1.x even maintained? No new version came out since 2.0 was published.
Will it be maintained in the future?
What when bitflags 3.x comes out?

Why not release defmt 2.0 that has an optional flags to use different versions of bitflags and reuse the trick from 0.3?
This way upgrades could be gradual.