knurling-rs/defmt #885: defmt-test: can't find crate for `test`

Gerharddc, 2024-11-15:
Hi, I am trying to set up defmt-test for unit testing with an app build using Embassy. I have added the following very simple test to my app's main.rs:

```
#[cfg(test)]
#[defmt_test::tests]
mod unit_tests {
    use defmt::assert;

    #[test]
    fn it_works() {
        assert!(true)
    }
}
```

but when I try `cargo test`, I am confronted with:

```
error[E0463]: can't find crate for `test`
```

Unfortunately I have no idea where this is coming from. I have tried to compare the basic of my project with the template at https://github.com/knurling-rs/app-template/tree/main, but cannot see any difference that would explain this issue.

I have the following Cargo.toml for my workspace:

```
[workspace]
members = ["bootloader", "app"]
resolver = "2"

# cargo build/run
[profile.dev]
codegen-units = 1
debug = 2
debug-assertions = true # <-
incremental = false
opt-level = 'z'         # <-
overflow-checks = true  # <-

# cargo test
[profile.test]
codegen-units = 1
debug = 2
debug-assertions = true # <-
incremental = false
opt-level = 3           # <-
overflow-checks = true  # <-

# cargo build/run --release
[profile.release]
codegen-units = 1
debug = 2
debug-assertions = false # <-
incremental = false
lto = 'fat'
opt-level = 3            # <-
overflow-checks = false  # <-

# cargo test --release
[profile.bench]
codegen-units = 1
debug = 2
debug-assertions = false # <-
incremental = false
lto = 'fat'
opt-level = 3            # <-
overflow-checks = false  # <-

[patch.crates-io]
embassy-stm32 = { git = "https://github.com/embassy-rs/embassy", rev = "d3fd34e5ffa45a71cca650fc13d2438f1c2f0861" }
embassy-futures = { git = "https://github.com/embassy-rs/embassy", rev = "d3fd34e5ffa45a71cca650fc13d2438f1c2f0861" }
embassy-net = { git = "https://github.com/embassy-rs/embassy", rev = "d3fd34e5ffa45a71cca650fc13d2438f1c2f0861" }
embassy-usb = { git = "https://github.com/embassy-rs/embassy", rev = "d3fd34e5ffa45a71cca650fc13d2438f1c2f0861" }
embassy-usb-dfu = { git = "https://github.com/embassy-rs/embassy", rev = "d3fd34e5ffa45a71cca650fc13d2438f1c2f0861" }
embassy-boot = { git = "https://github.com/embassy-rs/embassy", rev = "d3fd34e5ffa45a71cca650fc13d2438f1c2f0861" }
embassy-boot-stm32 = { git = "https://github.com/embassy-rs/embassy", rev = "d3fd34e5ffa45a71cca650fc13d2438f1c2f0861" }
embassy-sync = { git = "https://github.com/embassy-rs/embassy", rev = "d3fd34e5ffa45a71cca650fc13d2438f1c2f0861" }
embassy-time = { git = "https://github.com/embassy-rs/embassy", rev = "d3fd34e5ffa45a71cca650fc13d2438f1c2f0861" }
embassy-time-driver = { git = "https://github.com/embassy-rs/embassy", rev = "d3fd34e5ffa45a71cca650fc13d2438f1c2f0861" }
embassy-time-queue-driver = { git = "https://github.com/embassy-rs/embassy", rev = "d3fd34e5ffa45a71cca650fc13d2438f1c2f0861" }
embassy-embedded-hal = { git = "https://github.com/embassy-rs/embassy", rev = "d3fd34e5ffa45a71cca650fc13d2438f1c2f0861" }
```

And the following Cargo.toml for my app:

```
[package]
name = "mcu-fw-app"
version = "0.1.0"
edition = "2021"

[dependencies]
defmt = "0.3"
defmt-rtt = "0.4"

embassy-stm32 = { version = "0.1", features = [
    "stm32f439zi",
    "time-driver-any",
    "exti",
    "chrono",
] }
embassy-executor = { version = "0.6", features = [
    "task-arena-size-32768",
    "arch-cortex-m",
    "executor-thread",
    "executor-interrupt",
    "integrated-timers",
    "defmt",
] }
embassy-futures = "0.1"
embassy-boot-stm32 = "0.2"
embassy-time = { version = "0.3", features = ["tick-hz-32_768"] }
embassy-net = { version = "0.4", features = [
    "defmt",
    "tcp",
    "dhcpv4",
    "proto-ipv4",
    "medium-ethernet",
] }
embassy-usb = "0.3"
embassy-sync = "0.6"
embassy-embedded-hal = "0.2"

cortex-m = { version = "0.7", features = ["critical-section-single-core"] }
cortex-m-rt = "0.7"
embedded-io = { version = "0.6" }
embedded-io-async = { version = "0.6" }
panic-probe = { version = "0.3", features = ["print-defmt"] }
static_cell = { version = "2" }
picoserve = { version = "0.12", features = ["embassy", "defmt"] }

[dev-dependencies]
defmt-test = "0.3.2"
```

Any idea what could be causing this or advise for where I should be looking? Thanks!

jonathanpallant, 2024-11-19:
That error normally appears when you try and compile the standard Rust test runner for a bare-metal target that doesn't have the test crate available.

Have you got the extra `Cargo.toml` entries as described at https://crates.io/crates/defmt-test?

```toml
# Cargo.toml

# for the library crate (src/lib.rs)
[lib]
harness = false

# for each crate in the `tests` directory
[[test]]
name = "test-name" # tests/test-name.rs
harness = false

[[test]]
name = "second" # tests/second.rs
harness = false
```

Gerharddc, 2024-11-22:
@jonathanpallant oops, my bad, that was indeed the problem. Fixing that was indeed as simple as adding `test = false` to my bin in Cargo.toml so that I now have:

```
[package]
name = "mcu-fw-app"
version = "0.1.0"
edition = "2021"

[[bin]]
name = "mcu-fw-app"
test = false

...
```

Now I get other problems related to the cortex-m-rt linker script not finding the interrupt vectors it needs. Unfortunately I won't have time to troubleshoot that further as I have already moved on to using https://crates.io/crates/embedded-test and have that working. 

I had a similar problem with embedded-test though and in that case this was ultimately fixed by using their Embassy integration correctly. So for anyone struggling with that, maybe https://github.com/probe-rs/embedded-test/issues/48 can help.

closed 2024-11-22 by Gerharddc