knurling-rs/defmt #892: defmt-print and semihosting

andersesbensen, 2024-11-21:
Hi, I'm trying to get defmt working with qemu and the defmt-print tool, but I'm facing some issues.

In my test program I only have the lines 
    defmt::error!("Hello World");
    debug::exit(debug::EXIT_SUCCESS);
In my main routine

The problem is that why I try to decode the log stream with defmt-print, I always get a "malformed frame error"

qemu-system-arm -cpu cortex-m3 -machine stm32vldiscovery -nographic -semihosting-config enable=on,target=native -kernel qemu_test | defmt-print.exe -v -e qemu_test  stdin

(HOST) malformed frame skipped
└─ defmt-print @ C:\Users\ander\.cargo\registry\src\index.crates.io-6f17d22bba15001f\defmt-print-0.3.8\src/main.rs:145

The raw output of QEMU is 
$ qemu-system-arm -cpu cortex-m3 -machine stm32vldiscovery -nographic -semihosting-config enable=on,target=native -kernel target/thumbv7m-none-eabi/debug/test-qemu | xxd
00000000: 0100

If I feed the same stream to defmt-print I get the same error
$ echo -n -e "\x01\x00"| defmt-print.exe -e target/thumbv7m-none
-eabi/debug/test-qemu -v
(HOST) malformed frame skipped
└─ defmt-print @ C:\Users\ander\.cargo\registry\src\index.crates.io-6f17d22bba15001f\defmt-print-0.3.8\src/main.rs:145

Is this not supposed to work?

jonathanpallant, 2024-11-22:
What if you put a really long delay loop after the print and before the semihosting-exit? I wonder if it's a buffering issue.

jonathanpallant, 2024-11-22:
OK, I wrote an example, and it seems to work OK: https://github.com/ferrous-systems/rust-training/pull/228

```console
$ cargo run
    Finished `dev` profile [unoptimized + debuginfo] target(s) in 0.17s
     Running `/Users/jonathan/Documents/ferrous-systems/rust-training/example-code/qemu-thumbv7em/./qemu_run.sh target/thumbv7em-none-eabi/debug/qemu-thumbv7em`
ELF_BINARY=target/thumbv7em-none-eabi/debug/qemu-thumbv7em
Timer with period zero, disabling
Hello, world!
ERROR This is an error log
$ cat qemu_run.sh
#!/bin/bash

# This requires you to previously run `cargo install defmt-print`

ELF_BINARY=$1
echo "ELF_BINARY=$ELF_BINARY"
shift
qemu-system-arm -cpu cortex-m4 -machine lm3s6965evb -nographic -semihosting-config enable=on,target=native -kernel $ELF_BINARY | defmt-print -e $ELF_BINARY $*
$ qemu-system-arm -cpu cortex-m4 -machine lm3s6965evb -nographic -semihosting-config enable=on,target=native -kernel ./target/thumbv7e
m-none-eabi/debug/qemu-thumbv7em | xxd
Timer with period zero, disabling
00000000: 0001 7e00 027e 00                        ..~..~.
$
```

If you can publish your example as open source, we can take a quick look at it. If it's not open-source, Ferrous Systems offer professional support.

andersesbensen, 2024-11-22:
Thanks for quick reply. My example looks more or less identical to yours, except that I used a local logger implemetation, which I must have copied of an old branch or something. Because I used your current implementation it works perfectly, sorry my bad.

I think it would make sense to publish the defmt-semihosting crate to crates-io just like the defmt-rtt. The semihosting variant is very usefull in CI for other projects as well as defmt. The reason I came across this was that I would like to try to make a QEMU example for embassy-rs.

closed 2024-11-22 by andersesbensen