knurling-rs/defmt #898: Recent defmt-macros release broke "-Z direct-minimal-versions"

pdh11, 2024-11-28:
I build my project in CI with "-Z direct-minimal-versions", to check that I'm not accidentally using newer features of things than my Cargo.toml lists. Last night that CI failed, with lots of errors such as:

```
error[E0425]: cannot find function `acquire_header_and_release` in module `defmt::export`
   --> cotton-scsi/src/scsi_device.rs:6:38
    |
6   | #[cfg_attr(feature = "defmt", derive(defmt::Format))]
    |                                      ^^^^^^^^^^^^^
    |                                      |
    |                                      not found in `defmt::export`
    |                                      in this macro invocation
    |
   ::: /home/peter/.cargo/registry/src/index.crates.io-6f17d22bba15001f/defmt-macros-0.3.10/src/lib.rs:172:1
    |
172 | #[proc_macro_error]
    | ------------------- in this expansion of `defmt::unreachable!`
```

This appears to be because defmt-macros 0.3.10 -- which, given that version number, should really be semver-compatible with defmt-macros 0.3.9 -- is _not_ in fact compatible with defmt 0.3.8. (My crate depends directly on defmt 0.3.8, and only _indirectly_ on defmt-macros, so "-Z direct-minimal-versions" picks defmt 0.3.8 and defmt-macros 0.3.10.)

"cargo tree" excerpt before running "cargo +nightly update -Z direct-minimal-versions" (working):

> ├── defmt v0.3.8
> │   ├── bitflags v1.3.2
> │   └── defmt-macros v0.3.9 (proc-macro)
> │       ├── defmt-parser v0.3.4
> │       │   └── thiserror v1.0.69

"cargo tree" excerpt afterwards (compilation errors as above):

> ├── defmt v0.3.8
> │   ├── bitflags v1.3.2
> │   └── defmt-macros v0.3.10 (proc-macro)
> │       ├── defmt-parser v0.4.1
> │       │   └── thiserror v2.0.3

jonathanpallant, 2024-11-28:
Thank you for raising this. I'll get back to you shortly.

jonathanpallant, 2024-11-28:
Because you are effectively pinned to defmt-0.3.8, as you observe, you cannot use defmt-macros-0.3.10 because it calls APIs that were introduced in defmt-0.3.9. The mismatching of defmt and defmt-macro versions was not something we had ever considered.

For now, could you try specifying `defmt-macros = "=0.3.9"` as a dependency? That should stop the resolver selecting defmt-macros-0.3.10.

The other option is to upgrade to defmt-0.3.9. It is backwards compatible.

jonathanpallant, 2024-11-28:
I'm preparing a defmt-0.3.10 releases which pins the defmt-macros version at `=0.4.0` to avoid this problem in the future. See https://github.com/knurling-rs/defmt/pull/899.

I'm still considering whether to yank 0.3.9 or not.

jonathanpallant, 2024-11-29:
https://github.com/knurling-rs/defmt/releases/tag/defmt-v0.3.10 has been published, but I'm leaving this open (despite what Github thinks) until we've decided whether to yank 0.3.9 or not.

Please let me know if you're happy to add `defmt-macros = "=0.3.9"` to your `Cargo.toml` or if you'd prefer to see `defmt-0.3.9` yanked.

jonathanpallant, 2024-12-02:
defmt-0.3.9 and defmt-macros-0.3.10 are yanked.

closed 2024-12-02 by jonathanpallant