knurling-rs/defmt #904: The logo on docs.rs/defmt is missing

jonathanpallant, 2024-11-29:
You just get a missing-image box at docs.rs/defmt.

Urhengulas, 2024-12-02:
I am pretty sure the logo was shown before, so I think it might be a problem on the docs.rs side as well.

Urhengulas, 2024-12-02:
It is not docs.rs fault, but because the logo was fetched from the knurling website which not available anymore.

Urhengulas, 2024-12-02:
Instead of relying on an external link the image should be included locally.
- The image can be found at https://defmt.ferrous-systems.com/assets/knurling_logo_light_text.svg.
- The attribute is here: https://github.com/knurling-rs/defmt/blob/f5315e098129c372d32ed6fc0fe309d6302488ee/defmt/src/lib.rs#L15
- Documentation on the attribute is at https://doc.rust-lang.org/rustdoc/write-documentation/the-doc-attribute.html?highlight=html_#html_logo_url.
- To build the docs cd into `defmt/defmt/` (the `defmt` package, not the whole workspace) and run `cargo doc --open`. If it works the logo should appear in the top left.
- It should be tested that the image gets included in the archive produced by `cargo package`.

BenFordTytherington, 2024-12-06:
After lots of searching, and reading this issue: https://github.com/rust-lang/rust/issues/32104, I think using local images isn't possible / simple, and even by providing the url of the logo in the book folder I couldn't manage to render it with `cargo doc --open`

Urhengulas, 2024-12-19:
For me the logo shows when changing the link:

```diff
diff --git a/defmt/src/lib.rs b/defmt/src/lib.rs
index 605e96d..d4d4055 100644
--- a/defmt/src/lib.rs
+++ b/defmt/src/lib.rs
@@ -17,7 +17,7 @@
 
 #![cfg_attr(not(feature = "unstable-test"), no_std)]
 // NOTE if you change this URL you'll also need to update all other crates in this repo
-#![doc(html_logo_url = "https://knurling.ferrous-systems.com/knurling_logo_light_text.svg")]
+#![doc(html_logo_url = "https://defmt.ferrous-systems.com/assets/knurling_logo_light_text.svg")]
 #![warn(missing_docs)]
 
 #[cfg(feature = "alloc")]
```

Before:
[image]
After:
[image]