knurling-rs/defmt #925: :x for arrays with inferred format

jonathanpallant, 2025-01-20:
The manual says you can do this, and it works:

```rust
#[derive(defmt::Format)]
struct S { x: u8 }

let x = S { x: 42 };
defmt::info!("{}", x);    // -> INFO S { x: 42 }
defmt::info!("{:x}", x); // -> INFO S { x: 0x2a }
```

But this does not work, but it probably should:

```rust
#[derive(defmt::Format)]
struct S { x: [u8; 2] }

let x = S { x: [42, 255] };
defmt::info!("{}", x);    // -> INFO S { x: [42, 255] }
defmt::info!("{:x}", x); // -> I expect INFO S { x: [0x2a, 0xff] } but it's the same as before
```

Urhengulas, 2025-01-21:
I agree with you and would also expect this to work.

knoellle, 2026-03-16:
TL;DR: I investigated this and have come to the conclusion that it is solvable without changing either the way strings are printed (breaking change in the output format) or the wire-format.

### Details
There are two conflicting requirements:
1. Strings should be printed verbatim without quotation marks.
2. Structs should be printed similar to `fmt::Debug`, this includes strings being printed with quotation marks.

However, on the serialization side we cannot know ahead of time which display hint is supposed to be used, and no information about struct hierarchies is transmitted.
On the receiver/printer side, we do have information about the display hints, but don't know whether a particular string is inside a "struct".

Since #359, the derive macro uses the Debug display hint for all struct fields, overriding hint from an outer context.
I currently don't see a way to fix this but I'm open to suggestions.