knurling-rs/defmt #928: Allow renaming of `defmt` in `Format` derive

jamesmunns, 2025-01-30:
In the postcard ecosystem, I've started gating all deps behind versioned features, e.g. `defmt-v0_3` instead of defmt. I typically rename the package to allow multiple incompatible versions to exist.

I noticed that the `Format` macro does not like this:

```rust
#[cfg_attr(feature = "defmt-v0_3", derive(defmt_v0_3::Format))]
pub struct Key([u8; 8]);
```

leads to a compilation error:

```
error[E0433]: failed to resolve: use of undeclared crate or module `defmt`
  --> source/postcard-schema/src/key/mod.rs:26:43
   |
26 | #[cfg_attr(feature = "defmt-v0_3", derive(defmt_v0_3::Format))]
   |                                           ^^^^^^^^^^^^^^^^^^ use of undeclared crate or module `defmt`
   |
   = note: this error originates in the derive macro `defmt_v0_3::Format` (in Nightly builds, run with -Z macro-backtrace for more info)

For more information about this error, try `rustc --explain E0433`.
```

Serde [allows](https://github.com/serde-rs/serde/issues/1855#issuecomment-658485129) for something like:

```rust
#[derive(serde_other::Serialize, serde_other::Deserialize)]
#[serde(crate = "serde_other")]
struct Example { }
```

This might be nice to have for defmt as well.

jonathanpallant, 2025-01-30:
Oof. The expansion of the macro is relying on `defmt::_stuff()` working.

I guess we need that extra attribute to rename what it expects.

jonathanpallant, 2025-01-30:
Although it should be noted that we're going to a lot of trouble to make sure this kind of import naming will never be required for defmt, because 0.3 will have a semver-trick crate to import-export defmt-1.0.

closed 2025-03-20 by Urhengulas