knurling-rs/defmt #936: Use defmt with arbitrary transports (e.g. Wi-Fi, BLE, USB)

jasta, 2025-02-14:
TL;DR defmt's write! macro doesn't seem to let me write bytes to an arbitrary stream like `std::write!` does, limiting its usage to only debug probe use cases.

defmt seems like a really neat approach to getting more bang for your buck on tiny MCUs but unfortunately it doesn't seem to support a common use case of mine.  Specifically, I like to expose a kind of CLI-via-RPC mechanism that lets me replace the typical crude debug shell via the probe with a much more feature rich system that uses client tools to connect over any number of mediums like Wi-Fi, BLE, USB, etc and run command-line like programs on the MCU.  A common problem with this approach is that it adds a ton of storage for all the human readable strings and what-not that are only there for developer purposes, often ending up needing to trim things down to fit within my flash budget.  defmt could help a ton here, but in order to make that work I'd need a defmt equivalent of `std::write!` where I can defmt write into any byte stream.  All the usual caveats apply for defmt and my tooling will need to know what ELF binary I'm trying to talk to but this is easily achieved using all the same tooling that are required by probe-rs and other tools.

Is this something that can be easily supported?  Looking around at the internals I have some doubts this is straight forward but maybe I'm missing something...

jonathanpallant, 2025-02-14:
You could open a second RTT channel and use one for defmt and one for plain bytes. https://docs.rs/rtt-target/latest/rtt_target/ explains how.

jasta, 2025-02-26:
Hmm, I'm not sure how rtt_target helps here.  What I'm talking about is a way to write defmt encoded logs to arbitrary streams in a non global context.  For example, imagine that I wanted both logs to the "console" when connected to a probe _and_ a completely separate way to interact with the MCU via BLE that sent string like payloads back encoded with defmt as well (ie strings are in the elf binary but sent as numbers over the wire).  Special tools would be needed to interpret the BLE data stream just as they are with probe-rs but that the two systems aren't related to each other.  That is, it's not just a bifurcation of what is sent to the probe but rather completely different debug strings.

I understand this could be implemented by writing my own macro to replace defmt::write! that gives me the control to write to arbitrary data streams but I'm a little puzzled why we wouldn't want defmt supporting this use case natively, just like std:: write! does?

jonathanpallant, 2025-02-26:
OK, I think I misunderstood. I thought you were asking for a way to pass arbitrary byte steams over defmt. I now believe you are asking for a way of sending defmt logs over arbitrary byte streams.

defmt is configured to use a global logger and that won't change - most people don't want to choose where their defmt data goes on a log by log basis, and adding the argument would complicate things.

My solution would be for you to implement your own global defmt transport, and at the transport layer control where the bytes get routed to.

jasta, 2025-02-28:
Ok thanks, I think I'll go the route of just duplicating the defmt write macro and it's dependencies so that I basically have an alternate implementation that can sit alongside defmt proper.  I'll use the same encoding tho so that I can share host side decoder tools.  Feel free to close this issue as won't fix.

closed 2025-02-28 by jonathanpallant