knurling-rs/defmt #946: defmt-test: Rust Analyzers Run Test button does not work with defmt-test anymore

ckrenslehner, 2025-03-08:
Hi!

I was trying the [app template](https://github.com/knurling-rs/app-template) and could not run the test via the Rust Analyzer Run Test button. Running the tests via the CLI however worked. 

Output via CLI:
```
D:\Software\rust-testing\my-app> cargo test --test integration                                                                                                                                               03/08/25 18:33:09 PM
    Finished `test` profile [optimized + debuginfo] target(s) in 0.24s
     Running tests\integration.rs (target\thumbv7em-none-eabihf\debug\deps\integration-03bf9c8cdf8fa7e5)
 WARN probe_rs::config::registry: Found chip STM32WB55RGVx which matches given partial name STM32WB55RG. Consider specifying its full name.
 WARN probe_rs::config::registry: Matching STM32WB55RG based on wildcard. Consider specifying the chip as STM32WB55RGVx instead.
      Erasing ✔ [00:00:00] [########################################################################################################################################################] 12.00 KiB/12.00 KiB @ 69.38 KiB/s (eta 0s )
  Programming ✔ [00:00:00] [########################################################################################################################################################] 11.00 KiB/11.00 KiB @ 23.19 KiB/s (eta 0s )    Finished in 0.681s
<lvl> (1/1) running `it_works`...
└─ integration::tests::__defmt_test_entry @ tests\integration.rs:13
<lvl> all tests passed!
└─ integration::tests::__defmt_test_entry @ tests\integration.rs:8
D:\Software\rust-testing\my-app>
```

Output via Run Button:
```
Compiling my-app v0.1.0 (D:\Software\rust-testing\my-app)
    Finished `test` profile [optimized + debuginfo] target(s) in 0.84s
     Running tests\integration.rs (target\thumbv7em-none-eabihf\debug\deps\integration-03bf9c8cdf8fa7e5)
Error: probe-rs was invoked with arguments exclusive to test mode, but the binary does not contain embedded-test

Stack backtrace:
   0: <unknown>
   1: <unknown>
   2: hid_write
   3: <unknown>
   4: <unknown>
   5: <unknown>
   6: <unknown>
   7: <unknown>
   8: <unknown>
   9: hid_write
  10: BaseThreadInitThunk
  11: RtlUserThreadStart
error: test failed, to rerun pass `-p my-app --test integration`

Caused by:
  process didn't exit successfully: `probe-rs run --chip STM32WB55RG D:\Software\rust-testing\my-app\target\thumbv7em-none-eabihf\debug\deps\integration-03bf9c8cdf8fa7e5 tests --show-output` (exit code: 1)
```

Version:
```
D:\Software\rust-testing\my-app> probe-rs --version                                                                                                                                                          03/08/25 18:34:40 PMprobe-rs 0.24.0 (git commit: 6fc653a)
```

Greetings, Christian

Veykril, 2025-03-26:
rust-analyzer invokes tests with a name filter which probe-rs does not support

t-moe, 2025-06-25:
> [---] the binary **does not contain embedded-test**

Try [embedded-test](https://crates.io/crates/embedded-test), which works with rust-analyzer and integrates perfectly into your IDE. 

It can also handle the test args, that rust-analyzer prepends to the commandline.