knurling-rs/defmt #961: Log macros use relative `defmt` path instead of absolute

asasine, 2025-05-02:
The expanded log macros use `defmt::foo` instead of `::defmt::foo`, which can cause them to use any local module named `defmt` instead of the `defmt` crate as intended. [The Rust Reference on procedural macro hygiene](https://doc.rust-lang.org/reference/procedural-macros.html#procedural-macro-hygiene) recommends using absolute paths.

Veykril, 2025-05-02:
Changing that is unfortunately a breaking change (which we did by accident once and then reverted).

asasine, 2025-05-02:
I'd like to understand more about that. Does the project make a guarantee that the logging macros make that they'll respect the local `defmt` module? If not, I don't think this is truly a "breaking change" in the SemVer sense of the definition as there's no API change to the function-like proc macros themselves and any dependency on the old behavior was never official in the first place.

Veykril, 2025-05-02:
It broke for someone here https://github.com/knurling-rs/defmt/issues/953