knurling-rs/defmt #977: add note about `embedded-test` to `defmt-test`

rursprung, 2025-07-25:
the `defmt-test` documentation mentions [in one place](https://defmt.ferrous-systems.com/introduction?highlight=defmt-test#defmt-test-) that
> You might want to look at [embedded-test](https://crates.io/crates/embedded-test), which integrates a bit better with probe-rs.

however, this is not mentioned _anywhere_ else: not in the crate README, not on `crates.io` nor anywhere else in the docs.
if it is deprecated (or at least a better successor is available - which would still imply that it is deprecated for new projects?) it'd be great if this could be made visible in all these places (that would require a new patch release of `defmt-test` to be released so that it shows up on crates.io and docs.rs, etc.).

jonathanpallant, 2025-07-25:
We are the authors of defmt-test but have nothing to do with embedded-test, and whilst this crate is not highly active, we do use it on projects and it is not deprecated.

rursprung, 2025-07-26:
ah, ok. then it might make sense to reword the statement about integration with `probe-run`, `probe-rs`, etc.?
the place i quoted seemed to imply that it is deprecated (since `probe-run` is deprecated).