knurling-rs/defmt #992: Duplicate log, maybe a bug of probe-rs

fan2nd, 2025-09-21:
# Description
When I tried the serial loopback test, I sent '1', '2', '3', '4' in sequence. The data returned from the serial port had no issues, but there were duplicate entries in the defmt log. As shown in the figure below.

[image]

# Envirenment
- board: stm32u545re-nucleo
- software: embassy + defmt_rtt + probers
```
#![no_std]
#![no_main]

use defmt::*;
use embassy_executor::Spawner;
use embassy_stm32::{
    bind_interrupts, peripherals,
    usart::{self, Config, Uart},
};
use {defmt_rtt as _, panic_probe as _};

bind_interrupts!(struct Irqs{
    USART1 => usart::InterruptHandler<peripherals::USART1>;
});

#[embassy_executor::main]
async fn main(_spawner: Spawner) -> ! {
    let p = embassy_stm32::init(Default::default());
    info!("Hello World!");

    let config = Config::default();
    let mut uart = Uart::new(
        p.USART1,
        p.PA10,
        p.PA9,
        Irqs,
        p.GPDMA1_CH0,
        p.GPDMA1_CH1,
        config,
    )
    .unwrap();
    let mut buffer = [0u8; 32];
    loop {
        if let Ok(len) = uart.read_until_idle(&mut buffer).await {
            info!("{}", &buffer[0..len]);
            uart.write(&buffer[0..len]).await.unwrap()
        }
    }
}

```

jonathanpallant, 2025-09-21:
Can you also give us `probe-rs --version` and the versions of defmt and defmt-rtt in your lock file?

fan2nd, 2025-09-21:
You are right, this is a bug in embassy-stm32.

closed 2025-09-21 by fan2nd