knurling-rs/defmt #993: Some way to disable defmt if it is included as a transitive dependency

FeldrinH, 2025-09-25:
I have a dependency, which itself has a dependency on `defmt`. The trouble is that I don't use `defmt`, yet this transitive dependency causes `defmt` metadata (`_defmt_version_ = 4` symbol) to be included in the final build. As a result tools like probe-rs think that I am using `defmt` and compalin about a missing `.defmt` section.

It would be nice if there was some way to disable the `_defmt_version_` symbol if `defmt` is included as a dependency but not used. I'm not sure how exactly this could be implemented, but it would be really nice to have.

jonathanpallant, 2025-09-25:
Our recommendation is that crates put defmt behind a feature gate.

Which crate is pulling in defmt unconditionally?

FeldrinH, 2025-09-25:
> Which crate is pulling in defmt unconditionally?

https://github.com/David-OConnor/stm32-hal/issues/156

David-OConnor, 2025-09-25:
Concur with jonathan: This should be feature-gated.